pierrechevalier83/ferris #1: Recommended keymap

poltj18, 2020-07-22:
This keyboard looks great! Awesome work!

I'm new to QMK, layering, and small keyboards. Do you have recommended keymap for this board? I had a look for handwired/ferris in config.qmk.fm and didn't see one listed. I spent some time trying to put one together, but I don't feel like it is well optimized.

`keymap.json`
```
{"version":1,"notes":"","documentation":"\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n","keyboard":"handwired/ferris","keymap":"ferris_final_02","layout":"LAYOUT","layers":[["KC_Q","KC_W","KC_E","KC_R","KC_T","KC_Y","KC_U","KC_I","KC_O","KC_P","LGUI(KC_A)","LCTL_T(KC_S)","LALT(KC_D)","LT(1,KC_F)","LT(2,KC_G)","KC_H","KC_J","RALT_T(KC_K)","RCTL_T(KC_L)","RGUI_T(KC_SCLN)","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_N","KC_M","KC_COMM","KC_DOT","KC_SLSH","LT(4,KC_BSPC)","LSFT_T(KC_SPC)","RSFT_T(KC_ENT)","LT(3,KC_TAB)"],["KC_NO","KC_EQL","KC_ESC","RESET","KC_NO","KC_LCBR","KC_P7","KC_P8","KC_P9","KC_RCBR","KC_NO","KC_QUOT","KC_DQUO","KC_NO","KC_NO","KC_LPRN","KC_P4","KC_P5","KC_P6","KC_RPRN","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_P0","KC_P1","KC_P2","KC_P3","KC_BSLS","KC_TRNS","KC_TRNS","KC_UNDS","KC_TILD"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_LBRC","KC_AMPR","KC_ASTR","KC_NO","KC_RBRC","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_TRNS","KC_DLR","KC_PERC","KC_CIRC","KC_TRNS","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_EXLM","KC_AT","KC_HASH","KC_PIPE","KC_TRNS","KC_TRNS","KC_MINS","KC_GRV"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_F7","KC_F8","KC_F9","KC_F10","KC_LGUI","KC_LCTL","KC_LALT","KC_NO","KC_NO","KC_NO","KC_F4","KC_F5","KC_F6","KC_F11","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_F1","KC_F2","KC_F3","KC_F12","KC_TRNS","KC_TRNS","KC_NO","KC_NO"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_COPY","KC_UNDO","KC_NO","KC_FIND","KC_PSTE","KC_TRNS","KC_TRNS","KC_TRNS","KC_NO","KC_NO","KC_LEFT","KC_DOWN","KC_UP","KC_RGHT","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","RSFT_T(KC_ENT)","KC_TAB"]],"author":""}
```

precondition, 2020-07-27:
I am not Pierre, but allow me to chime in.

Since the Ferris is a columnar staggered board, I would recommend moving your LT(2) away from the G key and on the V key instead as it is easier to move fingers up and down than laterally.

The Ferris is also a tiny keyboard and you had the good idea to use mirrored home row mods but keep in mind that LAlt is (generally) not equivalent to Ralt aka AltGr.

Despite the easy access to Shift, you've refrained from having the need to activate a layer and shift at the same time, which is a good thing. However, it seems like you've skipped +=. You have to activate layer 1 and shift for "=". You also have no way to do `Ctrl`+`+` to zoom in.

Additionally, I can sense some misunderstanding concerning KC_TRNS (or the downward pointing triangle). It does not let through the keycode of current_layer­ - 1, it lets through the keycode of the layer below the current one in the **active layer stack**. If you activate layer 2 and try to hit "(", you'll actually get "h" because the layer 1 is not on the active layer stack. That also means that the transparent keys of your nav layer won't point to pure modifier keys, as they are laid out on layer 3, but on the home row mods of your base layer 0.

Speaking of the nav layer, I can't help but notice the distinct lack of any PGUP, PGDN, HOME, and END keys, do you never use those keys?

Finally, I can recommend replacing all your N/A keys by KC_TRNS as it would help for the times when you switch so quickly between layers while typing that you haven't fully released the layer key before typing the next letter.

pierrechevalier83, 2020-07-29:
Thanks for your interest, I still would like to do a bit of polish before opening a PR for it, but in the meantime, I've pushed my WIP keymaps to a branch on my fork of QMK.
You can find my current daily driver keymap (based on Workman with an RSTHD variant on layer 1 that I'm experimenting with) here: https://github.com/pierrechevalier83/qmk_firmware/tree/ferris_pierrec83_keymap/keyboards/handwired/ferris/keymaps/pierrec83
and my go at a "good default" here: https://github.com/pierrechevalier83/qmk_firmware/tree/ferris_pierrec83_keymap/keyboards/handwired/ferris/keymaps/default.

I do need to do a pretty render of the keymap and write a blogpost about it, but the key features are:
* Homerow modifiers and layers
* Symbol layers split over 2 layers where one hand holds down a homerow key and either the other hand or the same homerow or the other hand in one of the 10 most comfortable positions reaches for the symbol
* A Mouse layer (I have stopped using a separate mouse)
* A navigation layer with arrow keys and convenient ways to move things around in my window managers (ymmv on this one)
* A layer for function keys
* A layer for numbers with easier access to 0123 (on the homerow) as they're more common
* The highest layer is a bit of a convenience layer with enter, meta and conveniently placed keys for quick vim navigation (may need a bit of adapting for the qwerty based keymap)
* The thumbs never move away from their home position when typing as that was a source of typos for me. The non-homing thumbs are used for convenience shortcuts for me, such as changing workspaces, semi-maximizing windows, volume and brightness control... depending on the layer. For this one as well, some adaptation to one's own shortcuts would probably make sense.

pierrechevalier83, 2020-08-07:
FYI, I raised a PR to qmk with the "better default" keymap. Here is the write up with renders explaining the decisions. https://github.com/qmk/qmk_firmware/blob/94c75462148d1b9b33c2a7787c93edeacfc6fab4/keyboards/handwired/ferris/keymaps/default/readme.md

closed 2021-03-05 by pierrechevalier83