pierrechevalier83/ferris #2: Rust firmware

TeXitoi, 2020-07-27:
A Ferris keyboard that doesn't run pure rust, that't a sacrilege! :innocent:

precondition, 2020-07-27:
This is probably a good start : https://about.houqp.me/posts/rusty-c/

TeXitoi, 2020-07-27:
I'm biased, but https://github.com/TeXitoi/keyberon

pierrechevalier83, 2020-07-29:
Thanks for your interest, I definitely am interested in writing pure rust firmware for the Ferris eventually. The houqb way, while it's interesting doesn't seem like a good endgame to me as it really adds complexity on top of QMK for a tiny bit of abstraction gain.
I started by doing an atmega32u4 version for the Ferris for v0.1 as this was very well documented and it did result in a working keyboard, but this did limit my ability to write pure rust firmware for it as support for that chip is not yet in stable rust.

My understanding is that it will be easier to write pure rust for an arm version, which is one of the motivations for switching to arm in rev 0.2. In any case, I'll always offer qmk for those who want it as it's easy to write and is what everyone expects; but I will definitely look at keyberon for inspiration when is time to get rusty :smile:

TeXitoi, 2020-07-29:
STM32F0 is, IMHO, a good choice because of well supported rust crates and crystal less USB (but you need a LDO). Keyberon would work out of the box on it.

There is a crate for MCP23017 but not for PCA9539PW. It should not be too complicated to implement.

The only needed part to port a STM32F0 with a MCP23017 would be the multiple matrix handling, and it is quite straightforward as keyberon is more a library than a framework.

TeXitoi, 2020-09-04:
I see that you plan for a  Arm STM32F030F4P6TR. This MCU doesn't have USB. I think STM32F042F4P6 would be a better choice: crystal less USB, hardware DFU, and 32KB of flash (keyberon, in non optimised form with a bit more complex firmware (UART communication) is 20KB, so 16KB might be a bit small. It's a bit less cheap, but you don't need a crystal.

leshow, 2020-09-14:
This seems to be the most modern: https://gitlab.com/polymer-kb/firmware/polymer. It's based on https://gitlab.com/polymer-kb/firmware/keebrs

It uses async-await, and seems really well done.

borisfaure, 2023-04-16:
I wrote a rust firmware for the Ferris thanks to https://github.com/TeXitoi/keyberon. It is available at https://github.com/borisfaure/ferris-rs

TomzBench, 2023-04-16:
make sure to slap a big fat rust logo on it

pierrechevalier83, 2023-04-20:
> I wrote a rust firmware for the Ferris thanks to https://github.com/TeXitoi/keyberon. It is available at https://github.com/borisfaure/ferris-rs

Thank you so much for this, @borisfaure! I never ended up working on this myself, but I'm very glad it exists now thanks to you :)

closed 2023-04-20 by pierrechevalier83

tib888, 2025-02-21:
I also had the goal to write a rust firmware fom my nex keyboard, but then is accidentally found RMK - definitly worths a try: https://github.com/HaoboGu/rmk