pierrechevalier83/ferris #4: Where to buy a kit?

rmwphd, 2020-11-16:
Are kit purchases still available for this keyboard (esp. Compact, Bling, or Mini revs)? 

I couldn't find a website link in the Readme.md.

MangoIV, 2020-11-27:
The newer revs are still in the prototype my stage. I guess you gotta be patient :)

mreed20, 2020-12-06:
@rmwphd A kit would definitely be more convenient, but I found that it was actually pretty easy to order the parts from https://lcsc.com (using their "BOM Tool" to upload the bill of materials, `bom.csv`) and  PCB from https://jlcpcb.com (I uploaded the PCB schematic to the website and everything else was automatic).

rmwphd, 2020-12-07:
Hmm! Thanks for the tip, Michael! 

Nice last name btw ;-)

mreed20, 2020-12-07:
Many thanks, Reed :)

pierrechevalier83, 2020-12-17:
I still haven't got working firmware for my a prototypes, so I would advise for waiting it out or going with the previous arm revisions if you really want a kit now 🙂

rmwphd, 2020-12-18:
~We will watch your career with great interest.~ 

I'll wait patiently and maybe try kitting out an older rev for myself.

closed 2020-12-18 by rmwphd

pierrechevalier83, 2021-03-25:
Hi Reed,

I know it's been a long while, but if you're still interested, I just
started a group buy for the latest version of the Ferris keyboard.

Here is a form that you can fill out if you would like to get one. I plan
to ship orders in a month or so.

https://docs.google.com/forms/d/1T0Kk4zi-_mEqnTM6SoopiEEtbMQLUm-RHTR6Tu1Jl-o

All the best,
Pierre

rmwphd, 2021-03-25:
Thanks for the note, Pierre! I'll be sure to sign up :-)

MoSchaub, 2021-04-05:
I signed up in the form and I was wondering if there was any confirmation email because I didn't receive anything

pierrechevalier83, 2021-04-05:
Hi Moritz,

Thanks for registering your interest.

Sorry, there isn't a confirmation email. I'll be giving updates on slack
and by email as things progress.

All the best,
Pierre