pierrechevalier83/ferris #5: Cannot compile keymap.json with `qmk compile`

precondition, 2021-01-30:
This isn't an issue about the hardware per se but about the QMK firmware files.

```c
qmk compile ~/qmk_firmware/keyboards/ferris/keymaps/default/keymap.json
qmk compile ~/qmk_firmware/keyboards/ferris/keymaps/pierrec83/keymap.json
```
These two commands fail to compile the keymaps because the QMK CLI cannot find the keyboard files. The reason why the keyboard files cannot be found is that both of those json files have the following key:value pair.
```json
    "keyboard": "handwired/ferris"
```
This is incorrect. A correct value pair would be either of:
```json
    "keyboard": "ferris/0_1"
```
```json
    "keyboard": "ferris/sweep"
```

precondition, 2021-01-30:
I would have opened a PR on the qmk_firmware repo but I wasn't sure what to pick between sweep and 0.1

pierrechevalier83, 2021-03-05:
Thanks for the report, @precondition. Just noticed it now and raised a PR to qmk. Better late than never :)

pierrechevalier83, 2021-03-07:
The PR was merged upstream.

closed 2021-03-07 by pierrechevalier83