pierrechevalier83/ferris #12: Investigate missing elements from ibom.html generated by Github Actions

pierrechevalier83, 2021-04-14:
I've had a report that the `ibom.html` files generated by CI don't contain any parts.
Troubleshoot the issue and fix it :smiley:

laughedelic, 2021-05-22:
Are there any workarounds? I checked several releases and the latest that works for me is https://github.com/pierrechevalier83/ferris/releases/tag/ferris_release_0, I think. Later ones are empty. 

Unfortunately, that last working release is a bit different from the PCB I've got, so I'm trying to get the right version of the ibom.

I tried running it locally, but the result is the same.

pierrechevalier83, 2021-05-22:
Workaround: I host the boms for all latest boards at  https://cuddlykeyboards.com/ibom.
They don't render in all browsers, though so you may need to try more than one...

laughedelic, 2021-05-23:
Thanks! This helped me a lot. It worked fine in Chrome.