pierrechevalier83/ferris #13: Investigate IBOM on Safari not working

peauters, 2021-04-22:


DrMagPie, 2021-06-04:
Hi @pierrechevalier83,

I have encountered issue of iBOM not working for any browser.
issue lies in this regex check on lines 3020 - 3021
```js
  document.getElementById("github-link").innerHTML = "InteractiveHtmlBom&nbsp;" +
    /^v\d+\.\d+/.exec(pcbdata.ibom_version)[0];
}
```

problem is that `pcbdata.ibom_version` is `"ferris_release_4\n"`
but regex expects it to be something like this `"v2.3-50-g53ae\n"` 

so could you please fix it :)

pierrechevalier83, 2021-06-04:
Thanks @DrMagPie, this is a very helpful observation. Reading the generate_ibom_html code, I can see that some of the code assumes it is run from their own github repo and tries to pull metadata from git. This fails in the way I use it as I am using their repo as a submodule. I think a fix may be to modify the script upstream to allow overriding the version with a command line argument. Will discuss it upstream.