pierrechevalier83/ferris #17: V0.1 TRRS jack part discontinued

ankitp4t3l, 2021-09-05:
LCSC shows part number `C12569` as discontinued. I'm on the hunt for a replacement, but thought I'd leave this note in case anyone else has already found a replacement or for me to document what I find.

keyboard-magpie, 2021-09-05:
It's a commodity part, if you search by the more generic pj-320a name you'll find them in many places.

ankitp4t3l, 2021-09-05:
I see, I was a little concerned about the differences in spacing between

1. https://datasheet.lcsc.com/szlcsc/Made-in-China-PJ-320A_C12569.pdf
1. https://www.amazon.com/uxcell-Connector-Female-Socket-PJ-320A/dp/B07KY8NMD5

but in my comparing so far the max difference is <= 0.2mm; that's okay right? :D

ankitp4t3l, 2021-09-05:
Actually (without even looking at the PCB dimensions) I gathered that
```
left to right (port edge to pin or plastic nub) mm on first (top right) image on link 1:

| Link 1 | Link 2 |
|--------+--------|
|    3.7 |    3.5 |
|   5.15 |    5.3 |
|   8.15 |    8.3 |
|   10.7 |   10.5 |
|  12.15 |   12.3 |
|   13.5 |   13.6 |

pin "row" to pin "row" spacing from top view:

| Link 1 | Link 2 |
|--------+--------|
|    4.4 |    4.4 |

are the plastic nubs centered in both? sure looks like it
```

Largest difference appears to be the diameter of the plastic nubs (0.5mm difference), but I can deal with that if it's a problem (hopefully).

Going to place my order soon!

closed 2021-09-05 by ankitp4t3l