pierrechevalier83/ferris #18: Ordering presoldered boards from PCBway

555wolfman, 2021-09-27:
I love this keyboard but i am afraid that SMD soldering is out of my reach.

Therefore my idea is to use the assembly service from PCBway / JLCPCB / Elecrow

But handling with gerber files and BOM is also out of my reach.

so my question: are any other people interested in that and has someone the skills for ordering that in a way that there is a good chance that working boards will arrive?

I can act as a proxy for EU.

Thanks in advance

Olaf

bronzehedwick, 2021-12-08:
I am in a similar boat: I am infatuated with this keyboard, and would love to get my hands on one, but my soldering skills are at n00b level, and the thought of putting everything together shakes me to my core.

So more just agreeing, but have no useful skills to offer here :/

eth0net, 2022-12-13:
Did you guys ever get anywhere with this? Stuck in the same boat now myself :D

555wolfman, 2022-12-14:
> Did you guys ever get anywhere with this? Stuck in the same boat now myself :D

No. I found a group buy for „cornish-zen“ a similar design. I get mine delivered in januar after 1 year waiting :-(

eth0net, 2023-02-15:
> @eth0net do you still looking for a pcb? we made a group buy and we have a presoldered one that will be ready soonish in France.
> 
> The only thing that left to be soldered will be the switches (it’s the choc bling v 0.2)
> 
> 
> 
> are you interested? (where are you based?)

I am! Shall we take this to Discord / other messages? My details are on my profile 😃

555wolfman, 2023-02-16:
> @eth0net do you still looking for a pcb? we made a group buy and we have a presoldered one that will be ready soonish in France. The only thing that left to be soldered will be the switches (it’s the choc bling v 0.2)
> 
> are you interested? (where are you based?)

I am also interested. Located in EU.

julientaq, 2023-02-17:
Folks, the group sent the remaining pcb without telling me, so the last one went away. :/

What we can do is give you the price it costs us for the whole thing and maybe we could start asking folks around for a group buy. 

Wait before ordering anything as we discovered that the bom was not complete, we’ll update that in the coming week.