pierrechevalier83/ferris #23: Intermittently functional

bigtree908, 2022-03-24:
After assembling a Ferris 2.0 compact I found the ESD diode array got very hot. Only after removing the array did the keyboard work at all but only intermittently. I probably did something wrong. Do you have any troubleshooting suggestions?

pierrechevalier83, 2022-03-24:
In general, if the ESD diode array gets hot, it probably indicates a short
somewhere. I would use a voltmeter in continuity mode to try and detect any
unwanted short. Probably worth starting around the USB C connector as
that's the most fine pitched component.

Good luck with troubleshooting. Please let me know how it goes.

All the best,
Pierre

bigtree908, 2022-03-24:
I assembled a second with just one button, to test. I was supper careful and I've tested everything I can think of. Would there be a possible next cause?

pierrechevalier83, 2022-03-24:
There might be 2 separate issues: the ESD chip getting hot might have been
die to a short on it which went away when you unsoldered it.
In which case, the intermittent issue might be due to any other reason like
a defective processor chip or some other soldering problem.

In general, the first thing to troubleshoot is that power is correct
everywhere with a voltmeter. Maybe touch up any solder joints that look
less than perfect.

Can you describe the intermittent issue in more details? What exactly
happens?

bigtree908, 2022-03-24:
Ok, I think I figured out the ESD problem. I might be wrong, is it possible my usb-c socket switch the -+ data lines and that caused that issue?

I removed the ESD array. Now I have just the intermittent problem. I have just the left half built with just one button (the 0) key. When I press the button sometimes I get an '0' sometimes I don't. Sometimes I get many '0's.

When I assembled it I was careful to test each capacitor and resistor just before attaching. 
When I purchased the components I had to find alternatives for some. I tried to match their values best I could.
Here are the components I was unable to find an exact, part number, match on (other than resistors/capacitors). 
-Bidirectional diode
-Schottky barrier diode
-PolyFuse
-Dropout Regulator
-TVS diode array
-USB-C Jack

I really appreciate your help.

pierrechevalier83, 2022-03-24:
Ah. Thanks for describing the intermittent issue. I think that might be due to the processor polling for the Io expander from the right hand and not finding it.

You can work around it by removing the code that interacts with the right hand from the firmware for testing purposes or simply soldering the right hand and plugging it in.

If the + and - lines are mixed up, I would reflow the USB C connector as it's likely a short over there. If that doesn't do it, look for any other potential shorts along these traces.

pierrechevalier83, 2022-03-24:
Actually, + and - lines mixed up could also come from a component for which polarity matters being soldered in an incorrect orientation.

bigtree908, 2022-03-25:
Connecting the right side fixed the intermittent issue. However the right side is not responding at all. I'm going to try an audio cable that isn't spliced. Thanks for the help.

pierrechevalier83, 2022-03-25:
Just to be explicit, be sure to use a TRRS cable between the halves (with 4 poles). TRS cables that have only 3 poles aren't sufficient.

pierrechevalier83, 2022-10-11:
Also, I would advise against powering the keyboard in that state. It's
likely to date the components.