pierrechevalier83/ferris #24: About the qmk config: how does the layout macro work?

x4Cx58x54, 2022-05-05:
Hi! I found this project very interesting and I am trying to understand your qmk configuration code. You see, the `keymap` array has size `num_layer`x`MATRIX_ROWS`x`MATRIX_COLS`, that is, n x 8 x 10 in this project. However, the [`LAYOUT` macro](https://github.com/qmk/qmk_firmware/blob/8f660ca1e652b3b53580442d321642165adbc7d0/keyboards/ferris/0_2/0_2.h#L25) seems to disagree, as it has a different size. I doubt this definition will map the second row on the left side to the first row of the right side. Would you be so kind to explain why it works?

pierrechevalier83, 2022-05-05:
As far as qmk is concerned, the keyboard has 8 rows of 5 columns: 4 per keyboard half.

The layout macro describes how to map one list of keys which are layed out like the physical keys on the keyboard (could be seen as 3 rows of 10 + 1 row of 4) into the firmware's view of the keyboard.

Does that make more sense?

x4Cx58x54, 2022-05-06:
And that is to say, although 8x10 space is allocated for `keymap` array, half of it is used and the other half remains empty. I think I got it. Thanks!

pierrechevalier83, 2022-05-06:
Oh. I see. The surprising lines of code are this one: https://github.com/qmk/qmk_firmware/blob/8f660ca1e652b3b53580442d321642165adbc7d0/keyboards/ferris/0_2/config.h#L29 which defines MATRIX_COLS at 10 and this one: https://github.com/qmk/qmk_firmware/blob/8f660ca1e652b3b53580442d321642165adbc7d0/keyboards/ferris/0_2/config.h#L54 which allocates A1 to the 5 rightmost pins.

This does feel like an oversight now we've discussed it.
As well as the one where A1 is alloca