pierrechevalier83/ferris #25: Alternatives to the I/O Exapnder

cigani, 2022-05-26:
The backorder for the I/O expander is estimated at around 80 weeks ( https://lcsc.com/product-detail/I-O-Expanders_Microchip-Tech-MCP23017-E-SO_C47023.html ) // Are their alternative chips? I noticed the same issue with the ARM chip. The 71 ( Version that doesn't have its own crystal ) is available a bit of everywhere but the 72 used in the version 0.2 is a bit on the rare side. 

So are their alternatives for the I/O expander?

pierrechevalier83, 2022-05-26:
Hi,

I'm pretty sure I have some of the IO expanders left. Probably also have
PCBs. I'll confirm later today or tomorrow once I'm home and let you know.

All the best,
Pierre

cigani, 2022-05-26:
Hey if you have them and can ship to Europe -- I'll happily buy the stocks you have however you want to arrange the payment thanks a ton :) -- If not a company here is saying they "may" get the some of the expander stock in July but I think they're optimistic. The PCBs are also giving me some issues as well -- Uploaded the Gerber zip for the 0.2.compact release to JLCPCB and got a message back from them about some issues with the Gerber requiring additional cost

> Reason:
> 
> 1.The order info shows that there is only one design , but actually, there are 3 different designs in your file, so there will be an extra cost for it if you want to make it as show in your file.Please kindly choose the correct designs when you place your order next time.
> 
> 2.There are too many small slots in your file , it will take too much time to get it finished during the routing process.

Never got this back from them regarding other PCBs so I thought it was a curiosity // Wasn't sure if maybe the 0.2.compact Gerbers maybe had some strangeness in them?

pierrechevalier83, 2022-05-26:
The 3 designs is kind of correct: left hand, right hand, middle info card.
It's just already panelized. I'm not sure what the other comment is about.

cigani, 2022-05-28:
Hey any news on the PCB front? I think I managed to find an I/O expander from a fairly sketchy Alibaba seller but the PCB is still a blocker on my end // Is the problem they're mentioning about the breakaway tabs something I can edit with the gerber files? If so I'll just look into fixing it my self

pierrechevalier83, 2022-05-28:
> Hey any news on the PCB front? I think I managed to find an I/O expander from a fairly sketchy Alibaba seller but the PCB is still a blocker on my end // Is the problem they're mentioning about the [...]

Hi, yes, I have PCBs and IO expanders. Can you contact me here: https://www.cuddlykeyboards.com/contact-me so we can discuss details in private?

cigani, 2022-05-28:
Of course thanks mate.

cigani, 2022-06-05:
Hey I  sent you a message but haven't heard back. I assume you're busy but let me know when you have time to discuss

pierrechevalier83, 2022-06-06:
Hey, I replied to your messages that you sent with the "contact us" form on
my website, but it seems you're not receiving messages at the address you
used on the form. Can you double check?

iamlambda, 2023-01-17:
salut Pierre, je rencontre le meme probleme qu'OP: il n'y a plus aucun IO expander de dispo pour build ton ferris :/ Est-ce que le manque de pièce rend le projet mort? Est-ce que tu comptes faire une version avec 2 mcu pour simplifié et rendre à nouveau disponible la construction du ferris? Sur cuddly, tout est out of stock aussi :/