pierrechevalier83/ferris #30: Why the choice on TRRS cable ?

Velasko, 2023-06-30:
Checking at the schematics, seems like 2 pins are used for vcc and one for gnd.
Changing it to ( vcc | gnd | data | data ) would make it compatible with both trs and trrs cables and sockets.

quarthex, 2024-11-02:
I just checked the schematics.

Jack0: i2c_scl, i2c_sda, +3V3, GND
Jack1: i2c_scl_right, i2c_sda_right, VDD, GND

It looks good to me.

Velasko, 2024-11-04:
Oh, I see now.

I was actually looking into a fork of the project, which has a microcontroller on both ends, which allows that.
I suppose they kept that in order to change the original board as little as possible.

When I noticed that, I thought on creating the issue there, but noticed this also happened here. Thought it would be better to ask things here.

Thanks for the explanation and sorry for the inattentiveness.

Will close the issue as it has been clearly answered now.

closed 2024-11-04 by Velasko